ZZ-Cat/CRSFforArduino #12: Strange behaviour

wizz92, 2023-03-03:
I am using Arduino Uno, transmitter radiomaster tx 12 and receiver Radiomaster RP3.
95% times the library does not register changes in transmitter channels.
What can be an issue and how can I debug it?

ZZ-Cat, 2023-03-03:
> I am using Arduino Uno, transmitter radiomaster tx 12 and receiver Radiomaster RP3. 95% times the library does not register changes in transmitter channels. What can be an issue and how can I debug it?

The first thing I see here is your Arduino UNO is incompatible.
I clearly state this in the readme, on the front page.

[image]

I also have a list of development boards that CRSF for Arduino _is_ compatible with. Again, this is in the readme, on the front page.

[image]

If you have any one of the _compatible_ platforms listed above (preferably a Metro M4 Express), flash your code onto that, & see if the issue persists.
Also, if you wouldn't mind, can you post your code here, so I can see what's going on? TIA.

ZZ-Cat, 2023-03-17:
After two weeks of no response from the OP, I am closing this issue & moving on.

**To be clear:** I _will not_ be adding support for development boards such as the Arduino UNO & any other development board featuring any ATmega/AVR microcontrollers.

CRSF for Arduino, at bare minimum, is _compatible_ with SAMD21 microcontrollers (featured in development boards such as the Arduino Zero & Adafruit Metro M0 Express).

To keep up with the latest developments, I highly recommend using a development board that features a SAMD51 microcontroller (such as the Adafruit Metro M4 Express). Because this is the same as what I am using to develop CRSF for Arduino at the current time.

Any further issues that are submitted regarding CRSF for Arduino not working (or displaying "strange behaviour") on development boards such as the Arduino UNO/Nano, Arduino Micro/Leonardo, or Arduino Mega 2560 or any other board that I have listed on the main README as incompatible, these issues will be closed immediately. CRSF for Arduino is _not_ compatible with those boards.

closed 2023-03-17 by ZZ-Cat