ZZ-Cat/CRSFforArduino #36: Teensy 3.2, 3.5, 3.6, 4.0, 4.1

Kristabel28, 2023-09-14:
### Minimum requirements

- [X] Core clock speed is 48 MHz or faster?
- [X] Is the CPU an ARM Cortex M0+ or later?
- [X] Is the host microcontroller's UART peripheral able to support 420,000 (or higher) baud rate(s)?
- [X] Is your development board currently supported in the Arduino IDE?

### What development board would you like me to add?

PJRC Teensy 3.x, 4.x
Summary/Comparison table: https://www.pjrc.com/teensy/techspecs.html

Only Teensy 3.2 and 3.5 are 5V tolerant, but 3.x boards are being phased out. Definitely could just use a logic level converter to solve this issue though. 

All Teensy boards are supported in Arduino IDE: https://www.pjrc.com/teensy/td_download.html
The additional board manager URL is https://www.pjrc.com/teensy/package_teensy_index.json

I'm not too sure where to find information about the exact baud rate, I believe that can be set. More information here: https://www.pjrc.com/teensy/td_uart.html
Teensy 4 is definitely able to support the requested baud rate. "On Teensy 4 boards, the serial ports normally are limited to 6 Mbit/sec baud rate. However, maximum baud rate can be increased to 20 Mbit/sec by reconfiguring the UART clock source."

Teensy 3.2: 72 MHz, Cortex-M4
Teensy 3.5: 120 MHz, Cortex-M4F
Teensy 3.6: 180 MHz, Cortex-M4F
Teensy 4.0/4.1: 600 MHz, Cortex-M7

The Teensy is a very commonly used board when higher speeds are demanded. Hope you will be able to add compatibility for the board. Thank you very much.

Kristabel28, 2023-09-14:
I've just tested version 0.3.0 before the implementation of Compatibility Table, using a Teensy 3.5 with BetaFPV ELRS Lite RX, everything works perfectly. :) Hoping to update the compatibility table (I don't know how, perhaps if you could explain I can do it) and then test the GPS Telemetry (and maybe figure out how to do battery telemetry)!

ZZ-Cat, 2023-09-15:
> I've just tested version 0.3.0 before the implementation of Compatibility Table, using a Teensy 3.5 with BetaFPV ELRS Lite RX, everything works perfectly. :) Hoping to update the compatibility table (I don't know how, perhaps if you could explain I can do it) and then test the Telemetry function! 

Excellent. I am kinda on hiatus for about another two weeks. But, I _really_ want to get back into this as soon as I can.

IIRC, the Compatibility Table uses the predefined USB VID (Vendor ID) and PID (Product ID) for boards with native USB ports.

I can't say that I am familiar with the Teensy platform myself. However, if you would like to - help out with Teensy support (EG Testing), that'd be great.

Also, for now, the only telemetry that is currently available is GPS. I do have plans to add full telemetry feedback as time marches on. Basically, you will need to provide your own sensor modules and feed the data from that into CRSF for Arduino for telemetry.

Kristabel28, 2023-09-15:
I can help out with testing, sure! 

I've found the VID and PID for my T3.5, as follows, but I'm not too sure how to add them in. I have spent the whole afternoon looking through your code, but I am lost, sorry T^T

T3.5: 
USB_VID == 0x16C0
USB_PID == 0x0483

ZZ-Cat, 2023-09-15:
> I can help out with testing, sure!
> 
> I've found the VID and PID for my T3.5, as follows, but I'm not too sure how to add them in. I have spent the whole afternoon looking through your code, but I am lost, sorry T^T
> 
> T3.5: USB_VID == 0x16C0 USB_PID == 0x0483

Yea, I will need to do it "the old fashioned way" (IE using the board and chipset preprocessor defines to determine compatibility), because the USB VID & PID values aren't provided as preprocessor defines (like what Arduino and Adafruit do with their development boards).

Also, I will need to make some changes to the Uart constructor in the `DevBoards.cpp` hardware abstraction layer.
Because, _this_ 👇 happened when I tried to compile for your Teensy 3.5.
[image]

It's an interesting platform, for sure. I have been 50/50 on getting into it myself (after seeing Void Star Labs make extensive use of Teensy development boards in their projects).

ZZ-Cat, 2023-09-17:
> I can help out with testing, sure!
> 
> I've found the VID and PID for my T3.5, as follows, but I'm not too sure how to add them in. I have spent the whole afternoon looking through your code, but I am lost, sorry T^T
> 
> T3.5: USB_VID == 0x16C0 USB_PID == 0x0483

I tagged you over in #37, and resolved the compiler errors you see here.
That branch compiles well on Teensy 3.5 on my end. I am now waiting for you to test it and let me know how it goes.

closed 2023-10-09 by ZZ-Cat