ZZ-Cat/CRSFforArduino #52: Support for Raspberry Pico

PeterCasa, 2023-12-07:
### Minimum requirements

- [X] Core clock speed is 48 MHz or faster?
- [X] Is the CPU an ARM Cortex M0+ or later?
- [X] Is the host microcontroller's UART peripheral able to support 420,000 (or higher) baud rate(s)?
- [X] Is your development board currently supported in the Arduino IDE?

### What development board would you like me to add?

Would it be possible to add RP2040 support? eg Waveshare RP2040 Zero or any other RP2040 based DevBoard like the original Raspberry Pi Pico.
Thanks!

ZZ-Cat, 2023-12-07:
Dude, you read my mind.  
Abso-fricken-lutely!  
I was thinking of adding this at some point in 2024, and this can definitely be added to the list.

Also, it would be a good excuse for me to get my hands on a RP2040. =^/.~=

PeterCasa, 2023-12-10:
Great to hear! A library not supporting the RP2040 is a shame ;-)
I have taken a look at your code a bit and I think it shouldn't require much effort.
Thanks in advance! If you need someone for testing, I'm your man.

ZZ-Cat, 2023-12-10:
> Great to hear! A library not supporting the RP2040 is a shame ;-)
> I have taken a look at your code a bit and I think it shouldn't require much effort.
> Thanks in advance! If you need someone for testing, I'm your man.

Awesome. I'm on hiatus over the holiday season, right now.  

When the time comes for me to pick this up again in 2024, I will let you know.  
I'm still taking feedback throughout the holiday season, by the way.

ZZ-Cat, 2024-01-21:
@PeterCasa, I tagged you over in #72.  
Let me know how _that_ goes for you over there.

PeterCasa, 2024-01-26:
Great, thank you! Highly appreciated! I will give it a try as soon as possible.

ZZ-Cat, 2024-01-26:
You're welcome. =^/.^=

Also, be aware that because the RP2040 itself is still considered a "new" chipset (despite the fact that it's actually been around for a few years), PlatformIO themselves _only_ support the Arduino RP2040 Connect _and_ the official Raspberry Pi Pico RP2040 boards.  
Therefore, I could _only_ add those two to the Compatibility Table, because that's what's currently available.

In order for a board to _be_ supported by CRSF for Arduino, it `MUST` have support in both Arduino IDE _and_ PlatformIO.

If more RP2040 derivatives are supported by PlatformIO, I _can_ add them in as and when they become available. Until then... you're stuck with those two development boards.

closed 2024-02-09 by ZZ-Cat