ZZ-Cat/CRSFforArduino #55: CRSF for Arduino initialization failed error on ESP32

salamantos, 2023-12-26:
### Is there an existing issue for this bug?

- [X] I have searched the issues and there is no existing issue for this bug.

### What development environment are you using?

Arduino IDE

### What board are you using?

Adafruit Feather ESP32

### What part of CRSF for Arduino is this bug related to?

Other (please specify)

### Current behaviour

First of all thanks for supporting this libraray! I think this is great job!

I'm trying to get it running on my ESP-WROOM-32 dev board with Betafpv light 2400 RX. I'm using [this example](https://github.com/ZZ-Cat/CRSFforArduino/blob/Main-Trunk/examples/channels/channels.ino). I changed rx port to 25 and tx port to 27 according to pinout diagram (also did try 16 and 17 which are phusical pins of the board). But I'm getting this error: "CRSF for Arduino initialization failed".

So maybe I'm doing something wrong or this board is not supported yet?

[image]

### Expected behaviour

crsf.begin() returns true

### Steps to reproduce

My wiring 

[image]

salamantos, 2023-12-26:
UPD: I've change board in Arduino IDE from "ESP32 Dev Module" to "Adafruit ESP32 Feather" and now it works just fine!

But what is strange for me is that when I'm replacing [this line](https://github.com/ZZ-Cat/CRSFforArduino/blob/a00d47f2dab9574c1987b3214316239812bbe162/src/CRSFforArduino/src/Hardware/CompatibilityTable/CompatibilityTable.cpp#L93) with `device.type.devboard = DEVBOARD_ADAFRUIT_FEATHER_ESP32;` and select "ESP32 Dev Module" board, by chip starts constantly rebooting. So how do I add new board?)

ZZ-Cat, 2023-12-26:
It's likely that the library doesn't have support for your target development board yet.  
Plus, support for custom UART pins are not yet available.

I am aware that some ESP32 targets have different default UART ports and some UART ports somehow conflict with their native USB capabilities.

As for your development board rebooting, unpredictable things happen when you flash firmware that doesn't match the target that it's compiled for. I strongly recommend _against_ this practice.

> How do I add new board?

You wait for me to add it on my end, and then I will work with you in regards to testing it, to ensure it works in the way it's expected.  
Generally, there is a process where you request your development board to be added to the Compatibility Table by submitting an issue.  
Indirectly, you have already done this part.

The new board will need to be tested, to ensure that CRSF for Arduino is working correctly. Sometimes, it's not as simple as adding board definitions to the Compatibility Table, as there are some esoteric hardware-specific code that may be added in various places to accommodate the new board too.

For now, sit tight. I am currently on [hiatus](https://github.com/ZZ-Cat/CRSFforArduino/discussions/53) until January 14, 2024.

If you would like to tackle this yourself in the meantime, you _can_ fork the project; make your modifications; test it and ensure your changes are working; and then submit a Pull Request back here. Then, I will review it and work with you from there.

ZZ-Cat, 2024-02-20:
I am closing this for the reasons I have stated in #66.

If this issue is still persisting for them, they are more than welcome to open up a new issue, and we can sort things out from there.  
Not only that, I was unable to source the development board that they are using here. I found devices similar to it, but wasn't the same as the one described here.

closed 2024-02-20 by ZZ-Cat