ZZ-Cat/CRSFforArduino #59: Can I communicate and interact with TX?

zhuwei, 2024-01-03:
### Is there already an issue for this?

- [X] I have searched the issues and there is no existing issue for this feature request.

### Is your feature request related to a problem?

I want to create a simple control system instead of Opentx, Because many tx communicate in half-duplex and only have one GPIO connection, I don’t know whether CRSFforArduino currently supports it.

### Describe the solution you'd like

Use esp32 to communicate with tx, and then use mobile phone Bluetooth to communicate with esp32 for parameter configuration.

### Describe alternatives you've considered

There is currently no open source alternative

### Additional context

I just want to make a very simple remote control with no screen, only a switch button and voice broadcast, and the rest are set up through the mobile phone's Bluetooth. For example, a mini handle remote control or a car or boat remote control

ZZ-Cat, 2024-01-03:
Thank you for your suggestion. However, this is outside the scope of CRSF for Arduino.  
CRSF for Arduino aims to bring the [Crossfire RC Protocol](https://github.com/crsf-wg/crsf) to the Arduino ecosystem of development boards.

It is not an entire controller, transmitter, receiver, and flight controller suite of firmwares.  
It is simply a library for a widely used RC and telemetry protocol, and it's my attempt at bringing it to the Arduino development platform, while simultaneously supporting a diverse range of hardware targets.

CRSF for Arduino is intended to be used as the pipeline between a compatible CRSF (EG ExpressLRS or TBS Crossfire) receiver and your development board. IE Your development board is seen by the CRSF Protocol as the "Flight Controller".

One more thing... OpenTX is obsolete firmware, as EdgeTX has overtaken its place in the drone community. I wish you all the best in your endeavours, and the best of luck to you.

closed 2024-01-03 by ZZ-Cat

zhuwei, 2024-01-05:
I think it may not have been expressed clearly.
I want to use CRSF for Arduino and esp32 tx for data communication, just adding half duplex communication on the existing basis. Do you know if it is available now?

ZZ-Cat, 2024-01-05:
For a start, your question here _should not_ be in the Issues tab.  
Issues are for bug reports, feature requests and requests for new target development boards.  
Everything else goes to my [Discussions](https://github.com/ZZ-Cat/CRSFforArduino/discussions) tab.

If you would like more information, you can read the CRSF Protocol's documentation [here](https://github.com/crsf-wg/crsf/wiki).  
The physical UART for the CRSF Protocol used in CRSF for Arduino _is_ full duplex (and uses both UART Rx and Tx pins) and supports telemetry feedback, as of #47.  
As of #39, CRSF for Arduino _does_ support a handful of ESP32 targets.

What you _need_ to do in order to get up and running is you `MUST` have at least either a TBS Crossfire transmitter module and receiver, _or_ an ExpressLRS transmitter module and receiver. Either way, you connect your _receiver_ to your development board (in your case, your ESP32 target) via its default Serial1 UART port (usually this is marked as pins 0 for Rx and 1 for Tx); and you plug your transmitter module into the module bay of your handset (EG RadioMaster TX16S).  
Then, you flash your ESP32 _with_ CRSF for Arduino... preferably using one of the example firmwares to get you started and to see how the API works.

As for you making your own handset, that is outside the scope of CRSF for Arduino itself.