ZZ-Cat/CRSFforArduino #76: No such file or directory when compiling with PlatformIO and VsCode

berickson, 2024-01-25:
### Is there an existing issue for this bug?

- [X] I have searched the issues and there is no existing issue for this bug.

### What development environment are you using?

PlatformIO

### What board are you using?

Teensy 4.0

### What part of CRSF for Arduino is this bug related to?

Build/Compile/Upload

### Current behaviour

Compiling fails with "No such file or directory".  This looks very similar to #19, but now the issue is with PlatformIO / VsCode and not the Arduino IDE. 

```.pio/libdeps/blue/CRSFforArduino/src/CRSFforArduino/src/SerialReceiver/CRSF/CRSF.hpp:35:10: fatal error: SerialReceiver/CRC/CRC.hpp: No such file or directory
   35 | #include "SerialReceiver/CRC/CRC.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/blue/lib87d/CRSFforArduino/CRSFforArduino/src/CRSFforArduino.cpp.o] Error 1
.pio/libdeps/blue/CRSFforArduino/src/CRSFforArduino/src/Hardware/CompatibilityTable/CompatibilityTable.cpp:30:10: fatal error: ./CFA_Config.hpp: No such file or directory
   30 | #include "./CFA_Config.hpp"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/blue/lib87d/CRSFforArduino/CRSFforArduino/src/Hardware/CompatibilityTable/CompatibilityTable.cpp.o] Error 1
```

### Expected behaviour

No compiler errors.

### Steps to reproduce

Using VsCode with a PlatformIO project, Add https://github.com/ZZ-Cat/CRSFforArduino to lib_deps in platformio.ini. Compile.

### Additional information

I was able to "fix" the behavior for myself by creating a fork of the project at https://github.com/berickson/CRSFforArduinoFork where I #undef PLATFORMIO in several CPP files forcing it to use the same include paths that are used for Arduino. If I reference that repo from my platformio.ini libdeps, things compile fine.

ZZ-Cat, 2024-01-25:
Yeha. I noticed that in my CI, which compile tests all the boards in the Compatibility Table on PlatformIO, and it failed nearly _all_ of the Teensy boards for the same reason you're describing here.

What's interesting is... they all used to compile with impunity, until fairly recently. So, it looks more like something outside the scope of CRSF for Arduino changed and it has affected the compilation process for Teensy targets.

ZZ-Cat, 2024-01-26:
Well, I'll add this to #61 and go from there.

ZZ-Cat, 2024-01-26:
> Yeah, seems like path handling has changed in PlatformIO.

Okay, I looked into this on my end, and compile-tested the Main-Trunk:

- Version: 1.0.0
- Date: 2024-1-20
- Commit: c9563f10c6d1e8e14eaaef38a3465a2b673c0283

[image]

What's weird about this is, compiling the entire unified Teensy targets on my end succeeded.
The only thing I can think of is if you have a look at a253ab61cea4c36c3f34705dfc632a36779d876f, you will see that PlatformIO made changes to its path handling, and I originally had `$PROJECT_DIR` in CRSF for Arduino's `platformio.ini` configuration file.  
It looked a little something like this:

```
include_dir = $PROJECT_DIR/src/platformio/include
lib_dir = $PROJECT_DIR/src
src_dir = $PROJECT_DIR/examples/platformio
test_dir = $PROJECT_DIR/src/platformio/test
```

After a recent update to PlatformIO, that `$PROJECT_DIR` created a separate `$PROJECT_DIR` folder in CRSF for Arduino's directory and _that_ was what was causing the compilation process to fail.  
Since a253ab61cea4c36c3f34705dfc632a36779d876f, I updated the `platformio.ini` file to look a little more like this:

```
include_dir = src/platformio/include
lib_dir = src
src_dir = examples/platformio
test_dir = src/platformio/test
```

So, chances are, that on your end, there _may_ be a `$PROJECT_DIR` folder that may need to be deleted.  This _could_ be the cause of what's going on here.

ZZ-Cat, 2024-01-26:
Alright! I found it. =^/.^=

[image]

Now I know what I am dealing with here. Yea, _this_ is how I point the Arduino IDE to the rest of CRSF for Arduino's files, because the Arduino IDE doesn't like nested library files and folders. So, PlatformIO kinda has to come along for the ride.
It looks to me like I will need to point PlatformIO to the rest of CRSF for Arduino's files in a different way to Arduino IDE... which _is_ a fairly easy fix.

berickson, 2024-01-27:
I just checked and I didn't see any extra $PROJECT_DIR folder on my end, also I'm not specifying any folders in my platformio.ini, for what it's worth, here's what my platformio.ini looks like, nothing too special:

```
[platformio]

[env:blue]
platform = teensy
board = teensy40
framework = arduino
board_microros_transport = serial
lib_deps = 
	I2Cdevlib-Core
	I2Cdevlib-MPU9150
	https://github.com/micro-ROS/micro_ros_platformio
	https://github.com/berickson/Servo100Hz
	https://github.com/Seeed-Studio/Grove_3Axis_Digital_Compass_HMC5883L
	https://github.com/berickson/CRSFforArduinoFork
	#https://github.com/ZZ-Cat/CRSFforArduino
build_flags = -DUSB_SERIAL -DBLUE_CAR
upload_protocol = teensy-cli
board_microros_user_meta = microros.meta
monitor_speed = 921600
```

ZZ-Cat, 2024-01-27:
> I just checked and I didn't see any extra $PROJECT_DIR folder on my end, also I'm not specifying any folders in my platformio.ini, for what it's worth, here's what my platformio.ini looks like, [...]

I see. I dug a little more into the issue you're having here, and it seems to me like it's a lot more problematic than what I initially thought.  
So, fixing this is now my top priority.

By the way things are going, I may finish up doing another re-structure of the entire repository. 'Cause on my end, no matter how I tried to dick around with the includes, compilation still failed.  
A possible solution I can think of off the top of my head is to standardise the folder-and-file structure with Arduino's library specifications.  
Currently, CRSF for Arduino is set up as a PlatformIO _project,_ not a library. This is likely why we're getting compilation errors when CRSF for Arduino is used as a dependency... because I didn't initially test this part yet, and Version 1.0.0 is not officially released yet.

This bug isn't hardware-specific, by the way.  
I compile-tested this with my Adafruit Metro M4 Express and I got the same compilation errors as what you're getting.

closed 2024-02-05 by ZZ-Cat

berickson, 2024-02-14:
I was on vacation so I was just now able to test this out. I tried it out this morning by updating my platformio.ini library dependency with 
```
https://github.com/ZZ-Cat/CRSFforArduino.git @^1.0.0
```
and everything compiles and seems to work fine. Thanks!

ZZ-Cat, 2024-02-14:
Excellent. You're welcome. =^/.^=