ZZ-Cat/CRSFforArduino #104: Teensy 4.1 not working

christophers090, 2024-03-24:
### Is there an existing issue for this bug?

- [X] I have searched the issues and there is no existing issue for this bug.

### What development environment are you using?

Arduino IDE

### What board are you using?

Teensy 4.1

### What part of CRSF for Arduino is this bug related to?

Build/Compile/Upload

### Current behaviour

The rc_channels code on a teensy 4.1 uploads but only boot loops. It gets stuck on the crsf->begin(); line.

### Expected behaviour

Expect it to show the channels

### Steps to reproduce

Upload the code to a teensy 4.1

christophers090, 2024-03-24:
Taking a deeper look the const char *CompatibilityTable::getDevboardName() function in the compatibility table was returning an empty string. When I changed line 717 from return deviceNames[device.type.devboard]; to return deviceNames[device.type.devboard - 1]; I'm now getting "teensy4.1" returned as I should and everything works.

ZZ-Cat, 2024-03-24:
> Taking a deeper look the const char *CompatibilityTable::getDevboardName() function in the compatibility table was returning an empty string. When I changed line 717 from return [...]

That tells me I am missing an entry in the Compatibility Table, specifically with the `deviceNames[DEVBOARD_COUNT]` array.

Fortunately, that's an easy fix; and it can be pushed out in the next patch release to v1.0.x.
Thank you _very_ much for your work with this. You have saved me the time in figuring this one out, and now I can release a fix for this almost straight away.

ZZ-Cat, 2024-03-24:
I found the root cause of the problem!  
I was right. I _am_ missing an entry in the `deviceNames[DEVBOARD_COUNT]` array.  
In the screencap below, at [`line 184`](../blob/92200ea564ec9d3173dcadc7e9e8b7219a05293e/src/hal/CompatibilityTable/CompatibilityTable.hpp#L184), there is an index entry for the STM32F405RG in the `ct_devboards_e` enumerator.
[image]

Then, in the `deviceNames[DEVBOARD_COUNT]` array, the entry for the STM32F405RG is missing at [`line 368`](../blob/92200ea564ec9d3173dcadc7e9e8b7219a05293e/src/hal/CompatibilityTable/CompatibilityTable.hpp#L368).
[image]

Therefore, everything from the STM32F407VE entry all the way up to the Teensy 4.1 is "shifted down" one index value.

christophers090, 2024-03-24:
Nice! I'm happy it was an easy fix, and glad I could help! Thanks for this awesome lib, it saved me so much time!

ZZ-Cat, 2024-03-24:
You're very welcome. =^/.^=

If you're able, can you fork CRSFforArduino (if you have not done so already), and submit a Pull Request to the [Version-1.0.x-Maintenance-Branch](../tree/v1.0.x-Maintenance-Branch) with the fix that's outlined [here](104#issuecomment-2016941321)?

If you're doing the Pull Request thing, you `MUST` enable [Commit Signing](https://docs.github.com/en/authentication/managing-commit-signature-verification/about-commit-signature-verification) with either a [GPG Key](https://docs.github.com/en/authentication/managing-commit-signature-verification/about-commit-signature-verification#gpg-commit-signature-verification) or an [SSH Key](https://docs.github.com/en/authentication/managing-commit-signature-verification/about-commit-signature-verification#ssh-commit-signature-verification) _before_ the first commit in your Pull Request.  
If you already have this enabled, your commits will show up  as a green `Verified` tag next to each individual commit. Otherwise, it your commits will show up as a yellow `Unverified` tag next to the individual commit.  
Due to my branch protection rules, I simply _cannot_ accept Pull Requests with `Unverified` commits.

christophers090, 2024-03-25:
Will this just give me credit for the change?  I don't need credit but if there's some other reason that you want me to do this let me know.  I'm new to github and don't know how any of the commit signing or GPG key thing works. Thanks!

ZZ-Cat, 2024-03-25:
> Will this just give me credit for the change?  I don't need credit but if there's some other reason that you want me to do this let me know.  I'm new to github and don't know how any of the commit signing or GPG key thing works. Thanks!
> 

It's all good. =^/.^=

The only reason why I asked is 'cause I'm currently taking a break from this all, and if you wanted the fix done straight away, I can merge Pull Requests remotely.

I am able to fix this on my end, but it will need to wait until my break is over.

So, it's simply a question of "how quickly would you like this done"?

**Option A:** Add this yourself, and submit a Pull Request back to me. Then, I can cut-and-release v1.0.3 immediately after I merge the Pull Request, provided all of the checks (barring the Arduino IDE build check, 'cause I am currently having ongoing problems with that one) in the Pull Request pass.

Or...

**Option B:** Wait for me to start back on this on 2024-4-7 NZDT, and I will cut-and-release this fix as v1.0.3 within that week.

christophers090, 2024-03-25:
Ok I'll go with option B, it's working on my end, hopefully no one will run into this issue in the meantime. Have a fun break! 

Thanks

ZZ-Cat, 2024-03-25:
I will. =^/,..,^=

closed 2024-04-24 by ZZ-Cat