ZZ-Cat/CRSFforArduino #108: CRSF frame parsing timeout repeatedly expires thus dropping all frames

britannio, 2024-04-10:
### Is there an existing issue for this bug?

- [X] I have searched the issues and there is no existing issue for this bug.

### What development environment are you using?

PlatformIO

### What board are you using?

Raspberry Pi Pico RP2040

### What part of CRSF for Arduino is this bug related to?

RC Channels

### Current behaviour

When I set up the library to get rc channel data, the callback will run but the channel data never changes.
I only encounter this issue when my Pi Pico is connected to the [Adafruit PiCowbell CAN Bus](https://www.adafruit.com/product/5728).

### Expected behaviour

The rc data should update as normal.

### Steps to reproduce

I'm unsure how to reproduce this issue without the mentioned hardware.

### Additional information

Removing the following lines fixes the issue.
https://github.com/ZZ-Cat/CRSFforArduino/blob/Main-Trunk/src/SerialReceiver/CRSF/CRSF.cpp#L78-L87
So the frame expires before it is fully passed, causing `framePosition` to be reset to 0 mid-way through parsing a frame and thus the CRC check will fail.

britannio, 2024-04-10:
https://github.com/ZZ-Cat/CRSFforArduino/blob/26952d5351165cd8d6a42ac5d72a14b1c7eda686/src/SerialReceiver/CRSF/CRSF.cpp#L66

Increasing the timeout here may fix the issue but is the timeout necessary?

ZZ-Cat, 2024-04-12:
> ...is the timeout necessary?

Oh gods, you have my intrusive thoughts in a nutshell, here. :laughing:

The time-out is calculated based on the maximum possible size of CRSF packets, which is 64 bytes.  
From what I understand of the protocol, a timeout of approximately 1500 µS is required to receive all bytes _before_ the buffer is reset.  
If an incomplete packet is received, it is assumed that the connection between your receiver and your target (EG development board) is either janky or broken... at least _this_ is the theory behind it.

Removing the time-out altogether may fix _this_ issue. However, it creates more issues without it than it does with it...
It's more likely that my implementation _could_ do with a re-factor, as it _was_ derived from [Betaflight's](/betaflight/betaflight/blob/7c8a55258d069d7f531e42df72c11abc173e2c33/src/main/rx/crsf.c#L362) implementation.

> Increasing the timeout here may fix the issue but is the timeout necessary?
```cpp
timePerFrame = ((1000000 * packetCount) / (baudRate / (CRSF_FRAME_SIZE_MAX - 1)))
```

britannio, 2024-04-12:
> What do you mean by this? Could you elaborate on this a little more?

Each invocation of `CRSF::receiveFrames()` parses a single frame. The frame has a frame type and the ones accepted by the library are:
* `CRSF_FRAMETYPE_RC_CHANNELS_PACKED`
* `CRSF_FRAMETYPE_LINK_STATISTICS`

If the received frame is `RC_CHANNELS_PACKED` then the callback set in `CRSFforArduino::setRcChannelsCallback` should be invoked.
Likewise, if the frame is `LINK_STATISTICS` then the `CRSFforArduino::setLinkStatisticsCallback` callback should be invoked.
But right now the callbacks are being invoked irrespective of the frame type or whether the CRC check for the current frame passes.

If this change is made, the failsafe flag may need to be moved as it only relies on link statistics data.

ZZ-Cat, 2024-04-12:
> Each invocation of `CRSF::receiveFrames()` parses a single frame. The frame has a frame type and the ones accepted by the library are:
>
>     * `CRSF_FRAMETYPE_RC_CHANNELS_PACKED`
>
>     * [...]

Interesting. :thinking:
Yea, nah. It shouldn't be doing that, and I will look into it.
At this time, my focus is primarily with #103 (which has taken priority). So, it may be some time before I sort this issue out.

kallaspriit, 2025-03-17:
I can also confirm this on RP2040 with earlephilhower core.

Added some debugging:

```c++
/* Reset the frame position if the frame time has expired. */
if (currentTime - frameStartTime > timePerFrame)
{
    Serial.print("Timeout: ");
    Serial.print(currentTime - frameStartTime);
    Serial.print(" > ");
    Serial.print(timePerFrame);
    Serial.print(", start time:");
    Serial.print(frameStartTime);
    Serial.print(", current time:");
    Serial.println(currentTime);

    framePosition = 0;

    if (currentTime < frameStartTime)
    {
        frameStartTime = currentTime;
    }
}
```

And getting logs like:

```
Frame timeout: 1537 > 1512, start time:16301044, current time:16302581
Frame timeout: 15541 > 1512, start time:16302581, current time:16318122
Frame timeout: 1552 > 1512, start time:16318122, current time:16319674
Frame timeout: 1568 > 1512, start time:16319674, current time:16321242
Frame timeout: 23980 > 1512, start time:16321242, current time:16345222
Frame timeout: 1525 > 1512, start time:16345222, current time:16346747
Frame timeout: 1534 > 1512, start time:16346747, current time:16348281
```

The timeout logic in `CRSF::receiveFrames` does not make much sense to me. The `frameStartTime` should be set to `currentTime` when the `framePosition == 0` before the frame expire check? Otherwise is also counts time between frames? And when the frame time has expired then we should set `framePosition = 0;` and also `frameStartTime = currentTime;`?

Anyway when I changed the code to this the constant expiring problem disappeared.

```c++
bool CRSF::receiveFrames(uint8_t rxByte)
    {
        static uint8_t framePosition = 0;
        static uint32_t frameStartTime = 0;
        const uint32_t currentTime = micros();

        /* Reset the frame start time if the frame position is 0. */
        if (framePosition == 0)
        {
            frameStartTime = currentTime;
        }

        /* Reset the frame position if the frame time has expired. */
        if ((currentTime - frameStartTime) > timePerFrame)
        {
            framePosition = 0;
            frameStartTime = currentTime;
        }

        /* Assume the full frame length is 5 bytes until the frame length byte is received. */
        const int fullFrameLength = framePosition < 3 ? 5 : min(rxFrame.frame.frameLength + CRSF_FRAME_LENGTH_ADDRESS + CRSF_FRAME_LENGTH_FRAMELENGTH, (int)CRSF_FRAME_SIZE_MAX);

        if (framePosition < fullFrameLength)
        {
            /* Store the received byte in the frame buffer. */
            rxFrame.raw[framePosition] = rxByte;
            framePosition++;

            if (framePosition >= fullFrameLength)
            {
                /* Frame is complete, calculate the CRC and check if it is valid. */
                const uint8_t crc = calculateFrameCRC();

                if (crc == rxFrame.raw[fullFrameLength - 1])
                {
                    switch (rxFrame.frame.type)
                    {
                        case crsfProtocol::CRSF_FRAMETYPE_RC_CHANNELS_PACKED:
                            if (rxFrame.frame.deviceAddress == CRSF_ADDRESS_FLIGHT_CONTROLLER)
                            {
                                memcpy(&rcChannelsFrame, &rxFrame, CRSF_FRAME_SIZE_MAX);
                                rcFrameReceived = true;
                            }
                            break;

#if CRSF_LINK_STATISTICS_ENABLED > 0
                        case CRSF_FRAMETYPE_LINK_STATISTICS:
                            if ((rxFrame.frame.deviceAddress == CRSF_ADDRESS_FLIGHT_CONTROLLER) && (rxFrame.frame.frameLength == CRSF_FRAME_ORIGIN_DEST_SIZE + CRSF_FRAME_LINK_STATISTICS_PAYLOAD_SIZE))
                            {
                                crsf_payload_link_statistics_t linkStatisticsPayload;
                                memcpy(&linkStatisticsPayload, rxFrame.frame.payload, sizeof(crsf_payload_link_statistics_t));

                                linkStatistics.rssi = (linkStatisticsPayload.active_antenna ? linkStatisticsPayload.uplink_rssi_2 : linkStatisticsPayload.uplink_rssi_1);
                                linkStatistics.lqi = linkStatisticsPayload.uplink_link_quality;
                                linkStatistics.snr = linkStatisticsPayload.uplink_snr;
                                linkStatistics.tx_power = (linkStatisticsPayload.uplink_tx_power < 9) ? tx_power_table[linkStatisticsPayload.uplink_tx_power] : 0;
                            }
                            break;
#endif
                    }
                }

                /* Clear the frame buffer and reset the frame position. */
                memset(rxFrame.raw, 0, CRSF_FRAME_SIZE_MAX);
                framePosition = 0;
                return true;
            }
        }

        return false;
    }
```

But the rabbit hole goes deeper and this still fails in a more complex firmware with lots going on on both cores (I've got I2C devices, interrupts, other Uarts in use etc). I haven't been able to quite pinpoint what exactly is causing this but somehow one can't expect that reading from UART is always as fast and deterministic.

Somehow it seems to work fine without the timeout check, not sure in which condition would it break everything without it?

Also currently the timeout in CRSFforArduino is `1512`. In betaflight it's [1750](https://github.com/betaflight/betaflight/blob/master/src/main/rx/crsf.c#L54).

`#define CRSF_TIME_NEEDED_PER_FRAME_US   1750`

Increasing to `timePerFrame = 1750;` improves the situation but still getting quite a few.

Anyway food for thought, should take a harder look at the Betaflight code and compare. For now I guess I just have to include a modified version of the library in my codebase without the timeout check (or with relaxed timeout value) and hope stuff does not break in some conditions.

britannio, 2025-03-17:
https://github.com/britannio/pico_crsf may be useful.

kallaspriit, 2025-03-18:
I see You rolled out your own. Unfortunately it's not very Arduino-friendly. Also tried AlfredoCRSF which does work but the interface is not that great.

@ZZ-Cat do You think we're on to something here and have any time to take a look any time soon?

Also without creating a new issue, would You be open to supporting `std::function` based callbacks so we could use lamdas with capture and class member callbacks? Even if they're disabled by default and could be enabled with a define. C-style callbacks are limiting in a modern C++ environment and can't be used properly in OOP projects where you want a class instance member to handle the callbacks.

ZZ-Cat, 2025-03-18:
@kallaspriit Thank you very much for your insight into this issue. I greatly appreciate your help here.  

> The timeout logic in `CRSF::receiveFrames` does not make much sense to me. The `frameStartTime` should be set to `currentTime` when the `framePosition == 0` before the frame expire check? Otherwise [...]

Yea, that's exactly what should happen.  
I got the idea for the current timing logic from an earlier version of Betaflight and Rotorflight, as they both used the same logic at the time. If there is a better way of achieving that, then I'm all for it.

If what you have written here works reasonably well (barring the limitations outlined below), would you mind submitting it as a pull request? We can sort it out together from there.

> But the rabbit hole goes deeper and this still fails in a more complex firmware with lots going on on both cores (I've got I2C devices, interrupts, other Uarts in use etc). I haven't been able to [...]

That could be more to do with the hardware prioritising other interrrupts above what CFA is doing.
CFA's main function is called in your main loop, so interrupts are gonna take precedence over that. That could be why it would still fail in more complex systems.  
The alternative here would be to call the frame parsing in the same context as the UART's interrupt service routine and writing a bespoke UART driver for CFA - Essentially the same conclusion Betaflight developers came to. If I remember correctly, with both Betaflight and Rotorflight, they do _all_ CRSF frame parsing within interrupt context, and _this_ is likely the reason why.

Another possibility I would like to explore is using DMA to offload reading incoming UART bytes, so the incoming frames can be parsed a wee bit better instead of relying on the existing timing logic.

> would You be open to supporting `std::function` based callbacks so we could use lamdas with capture and class member callbacks? Even if they're disabled by default and could be enabled with a [...]

If [this](https://en.cppreference.com/w/cpp/utility/functional/function) is what you're referring to, then yes.  
I plan on phasing out all the C-style stuff that's present in CFA anyway, so that could be added to the list.

> Also currently the timeout in CRSFforArduino is 1512. In betaflight it's [1750](https://github.com/betaflight/betaflight/blob/master/src/main/rx/crsf.c#L54).

Yea. It seems they have updated that.  
When I first wrote the arithmetic for calculating that frame time-out, I did that based on what they had at the time, which was roughly around about the 1500 ms mark. This was also before [this](https://github.com/tbs-fpv/tbs-crsf-spec) was publicly available and I had to figure it all out on my own regarding how they arrived at that time-out figure. What made that even more dodgy was they chose an obscure baud-rate of 420K (because "lel 420 blaze it") which gave me a totally different figure than the one they were using at the time.  

> Anyway food for thought, should take a harder look at the Betaflight code and compare. For now I guess I just have to include a modified version of the library in my codebase without the timeout check (or with relaxed timeout value) and hope stuff does not break in some conditions.
> 
> Maybe You can test this and add a counter for these expire events if logging to console might be causing too much delay by itself and see how many of these events are you getting normally. Not sure if this is just a RP2040 issue or a more general problem but it seems that lots of valid frames get thrown away.

That would be worthwhile looking into.

> You think we're on to something here and have any time to take a look any time soon?

It looks like you certainly are.  
I could take a look at it over the next two weeks (no, I'm not perpetuating _that_ "Two Weeks:tm:" joke). I recently acquired a couple of RP2040-based development boards for investigating this specific issue.

---

> Also tried AlfredoCRSF which does work but the interface is not that great.

Heheheh. That's a +1 for CFA. =^/,..,^=  
No seriously, my primary goal for CFA is for it to be an Arduino-compatible implementation of CRSF in its entirety, and keep it actively maintained as the protocol itself evolves (EG CRSF v2 to CRSF v3).  
There have been folks out there that have attempted it before me, and kudos to them for stepping up to the plate and doing it. With that being said, it seems a lot of CFA's predecessors focused on implementing a small part of the protocol (usually either RC channels _or_ telemetry) and once that was done, they would leave their repositories largely unmaintained (which can create its own raft of problems).

ZZ-Cat, 2025-03-21:
> I could make a PR if it helps or do You plan to overhaul this yourself anyway?

That would be a welcome help in the interim.  
I may not re-factor the code-base straight away, as I currently believe doing so would merely create more work for me than what should be necessary.

> Doing the frame parsing in interrupt context sounds reasonable I guess, worth a try.

A re-factor may be in the cards at some point down the line. Up until then, CFA currently has no drivers of its own, as it relies on Arduino's API. I prefer to _not_ reinvent the wheel as much as I can. However, if it gets to the point where writing my own UART driver for CFA becomes unavoidable, rest assured it _shall_ be done.  
I have a fair idea of how that could be achieved: I am aware the bulk of CFA's compatible hardware supports DMA, and a handful of hardware also supports hardware-accelerated events. Both of these could be implemented in CFA at some point down the line, and that could mitigate the need for reliance on both the current implementation _and_ relying on timely interrupt execution. Either way, these are concepts I would like to explore if a re-factor is needed.

> In the protocol description there is a sync byte (with several options in the full spec), can that be considered a unique frame header?

[Here](https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#frame-construction), they don't specify the sync byte as a unique frame header. Yet [this](https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#device-addresses) seems to be it.  
They aren't clear on the purpose of the "sync byte" either, which doesn't help.

> One could just build a little state machine where we wait for the frame header, read the byte stream until valid frame is found (or not found) and not really care about tight timing.

That could be possible.  
The current implementation already assumes an arbitrary frame length until the _actual_ frame length byte is received.  
My understanding of the time-out thing is in case of either a physical hardware disconnect _or_ the data stream is somehow interrupted by a fail-safe, and then reconnected, the stream can carry on as though nothing has happened once the connection is re-established. I don't know. I have not been able to reproduce this, because it's some really obscure edge case scenario that the Betaflight developers solved aeons ago, and I used that for the timing with CFA.  
However, it seems the current timing implementation is more of a hindrance than a help, so it could do with not being there at all. Especially when it's there to solve a problem what has a smaller than 🤏 _that much_ of a chance of happening in reality; a better implementation could be written - For example, your suggestion.

Ultimately, I do not know, and I don't have an opinion on it other than "if it can be improved upon, by all means go for it".  
'Cause that's a part of what CFA is all about - continuous improvement.