ZZ-Cat/CRSFforArduino #109: Raspberry Pi Pico W

britannio, 2024-04-10:
### Minimum requirements

- [X] Core clock speed is 48 MHz or faster?
- [X] Is the CPU an ARM Cortex M0+ or later?
- [X] Does the host microcontroller have at least 32 KB of flash memory?
- [X] Is the host microcontroller's UART peripheral able to support 420,000 (or higher) baud rate(s)?
- [X] Is your development board currently supported in the Arduino IDE?

### What development board would you like me to add?

It seems like this is the only change to make: https://github.com/britannio/CRSFforArduino/commit/d699599c98fd81b6892564f6d36f5fbb35719b60

ZZ-Cat, 2024-04-12:
# Issue closed: Pico W not supported in [PlatformIO](https://registry.platformio.org/platforms/platformio/raspberrypi/boards)

You have reminded me that I need to update my issue template here and clarify that a potential target MUST be supported in _both_ the Arduino IDE _and_ PlatformIO.

The Pico W, IIRC uses [Earle Philhower's](https://github.com/earlephilhower) [Arduino Pico](https://github.com/earlephilhower/arduino-pico) package, of which is _not_ supported by PlatformIO at this time.  
This means that, for the time being, it is available _exclusively_ in the Arduino IDE. Thus, PlatormIO users lose out.

I _highly_ recommend you make some noise in [this Issue](https://github.com/platformio/platform-raspberrypi/issues/13) over on PlatformIO's repository, because I simply _cannot_ add more Pico-based development boards beyond what is simultaneously supported in the Arduino IDE _and_ PlatformIO.

ZZ-Cat, 2024-04-13:
> It's not officially supported by PlatformIO

The focus is to add boards that are _officially_ supported in the Arduino IDE (either by Arduino themselves or through third party board packages via the `Additional Boards URL` field) _and_ PlatformIO.  

I simply will not add boards with unofficial support in both the Arduino IDE or PlatformIO to CRSF for Arduino's Compatibility Table. If the board-in-question is _officially_ supported in _both_ the Arduno IDE _and_ PlatformIO, _then_ I will add it to the Compatibility Table.

Adding boards with unofficial support, or adding boards that are supported in _one_ IDE and _not_ the other creates a level of uncertainty and inconsistency with CRSF for Arduino that I am not willing to take on. The juice is not worth the squeeze.

---
> it is quite trivial to make it work

I understand that it's easy to implement your request. Be that as it may, simplicity versus complexity does not guarantee your request will be fulfilled, especially when there is a valid reason as to why it cannot be fulfilled.  

---

> this is how I'm using the Pico W so PlatformIO users do not lose out.

I sympathise greatly with folks that are using RP2040-based development boards and their inability to use them with PlatformIO beyond the two boards that are _officially_ supported. I would add support for every RP2040-based target in a heartbeat if they _were_ officially supported in PlatformIO. However, at this time (and for the forseeable future), they are not; and this _is_ something that is beyond my control. Therefore it is _not_ my call to make.

Rest assured, as _soon_ as PlatformIO _officially_ supports most RP2040-based targets, they _will_ be added to the Compatibility Table here in CRSF for Arduino.

My decision is final, and this is _not_ up for debate.  
I am locking this discussion, because it is finished. I am calling it here, and moving on.

ZZ-Cat, 2024-04-14:
Hi @britannio 
Apologies for being too abrupt yesterday, and hastily shutting this Issue down.

I was reading through [this section](https://arduino-pico.readthedocs.io/en/latest/platformio.html) of Earle Philhower's [Arduino-Pico](https://github.com/earlephilhower/arduino-pico) package, and it turns out I was wrong. My information on why the vast majority of RP2040-based targets couldn't be added to CRSF for Arduino's Compatibility Table was outdated.  
Based on the new information I was reading today, it actually _can_ be done. Looks like you and a lot of other folks will get more RP2040 support after all.

So, I have re-opened this issue and removed the "I'm not doing this" label and replaced it with the "In progress" label.

As I am writing this comment, I am currently checking out the Arduino-Pico package, and I am conducting some internal tests in a private branch, to ensure everything is all a-okay. Once the build succeeds, I will publish the branch and attach a Pull Request to it, and go from there.

britannio, 2024-04-15:
:)