ZZ-Cat/CRSFforArduino #111: Version 1.1.0 Development Tracking

ZZ-Cat, 2024-04-12:
I decided to create an issue for you all to see my priority list with development of Version 1.1.0, for the sake of transparency.  
There is a [milestone](../milestone/5) for Version 1.1.0, and you can keep up with my [development schedule](https://github.com/users/ZZ-Cat/projects/7). However, this issue serves as a quick-glance central point-of-interest for you all to see where I am currently at with everything.

The list is in order of priority, where the top-most item is the highest priority (where I am focusing the majority of my development time), and the lowermost item is the lowest priority (IE A significant amount of time may pass before I get around to working on this, while I make my way through higher priority tasks).

The list is as follows:

- [ ] #107
- [ ] #103
- [ ] Update Contribution Guidelines
  - [ ] Emphasise commit signature verification
  - [ ] Pull Requests
  - [ ] State when/how/why an Issue or Pull Request is rejected, and what to do when it happens
  - [ ] Update Do's and Don'ts
  - [ ] Remove unnecessary details
- [ ] Add PlatformIO requirements to Issue Templates
- [ ] Clean-up
  - [ ] Conduct Static Code Analysis and fix any errors this calls me out for
  - [ ] Remove obsolete Arduino IDE CI
  - [ ] Consolidate remaining CI automations and call it "Quality Control" or "QC" (actual name is yet to be determined)
- [ ] Add Static Code Analysis to consolidated CI
- [ ] #89
- [ ] #108

Anything else that comes up during development will be added to this list and appropriately prioritised.

## Additional

During this time, I ask that you _please_ be patient and understand that I am but _one_ person working on CRSF for Arduino.  
I have had my fair share of contributors as time has marched on, and they are not without my respect. I am very grateful for their efforts, and my contributors have been a huge help in lightening the load for me.

However, 98% of the time I am the _only one_ working on CRSF for Arduino, and as its popularity grows, I need to prioritise my focus in order to preserve my work-life balance, negate burning myself out _and_ to negate myself from being overwhelmed with a perceived "MUST do ALL the things... AT ONCE!!!!1111!!!11!"

If you are able, I encourage you to get involved with my project in the following ways:

- Testing the code base on supported hardware;
- Bug-testing and submitting feedback;
- Pull Requests;

> [!NOTE]
> For Pull Requests, you `MUST` have either GPG or SSH signing enabled _before_ your _first_ commit to your Pull Request's branch. All commits in your Pull Request `MUST` have the green `verified` badge next to them.  
> This is the same reason why you write your signature on an official document - It verifies that it's _you_ that wrote the commit and _not_ someone trying to impersonate you.

BadPotato1007, 2024-04-30:
So do you have an approximate of when this functionality will be ready? I am planning to use this lib to make a fully diy handset from scratch that can use elrs external modules...

ZZ-Cat, 2024-04-30:
> So do you have an approximate of when this functionality will be ready?

As my dad used to say: "It will be ready when it's ready." =^/.~=  
Or, as Trappy of Team BlackSheep would occasionally joke: "Two weeks."

For real, though. I don't do the whole "deadline" thing.  
I'd rather get this done right the first time, as opposed to rushing to meet ~~unrealistic~~ deadlines.

It's unrealistic to ask for a specific "due by" date, when the very nature of software development is unpredictable at best and has its fair share of delays and setbacks at worst.

Another thing ya gotta remember is... I'm just a gal what does this in her free time.  
I'm not funded by some big corporation or employed to do this "full-time". I write this stuff on my own terms.