ZZ-Cat/CRSFforArduino #140: Please help a novice. No such file or directory  #include "cstring"

VicAMMON, 2024-09-25:
### Is there an existing issue for this bug?

- [X] I have searched the issues and there is no existing issue for this bug.

### What development environment are you using?

Arduino IDE

### What board are you using?

Arduino Nano 33 IoT

### What part of CRSF for Arduino is this bug related to?

Build/Compile/Upload

### Current behaviour

SerialBuffer.cpp:26:10: fatal error: cstring: No such file or directory
 #include "cstring"

### Expected behaviour

expected all works fine

### Steps to reproduce

Just add library to Arduino IDE 2.3.2, and try to compile example.

nickdundee08, 2024-10-24:
I had the same problem and stumbled on a fix by replacing #include "cstring" with #include "string.h" in serialbuffer.cpp

ZZ-Cat, 2024-10-29:
`cstring` is a C++ header, and `string.h` is a C header.  
As a temporary workaround, you MAY use `string.h` in place of `cstring`. However, this is NOT RECOMMENDED because CRSF for Arduino is written in C++ (specifically C++23), _not_ C.

I am not going to fix this straight away, as I have higher priorities with CRSF for Arduino.  
**For example:** As my work with #138 progresses, this issue could resolve itself on its own while I clean up the code-base.  

Currently, I have taken time off of all of my public facing projects and I will not be back until early 2025. In the meantime, I may check-in from time-to-time, and triage issues, but that's it for the time being.

**For future reference:** The Issues tab is reserved for bug reports and requests for new hardware support. For requests for help (among other things), use the Discussions tab.