ZZ-Cat/CRSFforArduino #143: Adafruit M4 Express

todd-trowbridge, 2024-12-08:
### Minimum requirements

- [X] Core clock speed is 48 MHz or faster?
- [X] Is the CPU an ARM Cortex M0+ or later?
- [X] Does the host microcontroller have at least 32 KB of flash memory?
- [X] Is the host microcontroller's UART peripheral able to support 420,000 (or higher) baud rate(s)?
- [X] Is your development board currently supported in the Arduino IDE?

### What development board would you like me to add?

Adafruit M4 Express is showing up in hardware ID's as 8022 not 8020 as shows in Device Manager > Hardware ID of board

ZZ-Cat, 2024-12-10:
This really should be a bug report, not a request for a new development board.  
There is already several Adafruit M4 targets available.

Also, which "M4 Express" are you referring to?  
Because there are two possible "M4 Express" boards you could be referring to here:

- [Feather M4 Express](https://www.adafruit.com/product/3857)
- [Metro M4 Express](https://www.adafruit.com/product/3382)

Which "M4 Express" target are you referring to?

ZZ-Cat, 2025-03-21:
I am closing this issue because the OP did not get back to me within a reasonable time-frame when asked to specify what they were referring to.

closed 2025-03-21 by ZZ-Cat