ZZ-Cat/CRSFforArduino #146: Extended FRAMES - Unable to reach FC

ansl, 2024-12-30:
### Is there already an issue for this?

- [X] I have searched the issues and there is no existing issue for this feature request.

### Is your feature request related to a problem?

With a Lua Script I am trying to send a ping Frame (0x28) to the FC (ESP32 S3) but it does not reach the FC (I am monitoring the serial port and only frames 0x14/0x16 are delivered). I have tried broadcasting the ping and I get answers from the XF Micro TX, XF Nano RX V3 and XF Wifi.

### Describe the solution you'd like

I would like to be able to receive in the FC extended Frames

### Describe alternatives you've considered

I have modified CRSF.CPP to print the raw received frames, and I have tried sending all frames available in the documentation. Only frames 14 and 16 are delivered. It seems the receiver it is not forwarding the frames over CRSF TX/RX outputs. I honestly don't know what else to do, any help will be more than welcomed.

### Additional context

I have a TX16s(Edgetx) with a XF Micro TX(Baudrate:400k) and a XF Nano RX V3 as a receiver. The receiver (CRSF TX/RX) is connected to and FC (ESP32 S3) serial port(Baudrate:416666). I am receiving in the FC the frames 14 and 16 without any problem. Telemetry is enabled and I am successfully sending telemetry frames to the RC.

ZZ-Cat, 2024-12-30:
Okay, cool.  
Support for extended frames is currently not yet available. IIRC, extended frames are a part of CRSF V3 and CFA is currently compatible with a small portion of CRSF V2.

I do plan to re-factor CRSF for Arduino in the near future to bring it inline with CRSF V3, now that the [official documentation](https://github.com/tbs-fpv/tbs-crsf-spec) exists.  
There is a _lot_ I need to implement, as far as that is concerned.

Thank you for bringing this to my attention.  
However, it will be quite some time before I get around to fulfilling this Issue as there is a lot on my plate at this time.

ansl, 2024-12-30:
Thanks a lot for the fast response. You have here a cool piece of work! Thanks a lot for the effort. 

I'll stay tuned for the V3 implementations

/Antonio

ZZ-Cat, 2024-12-30:
> Thanks a lot for the fast response. You have here a cool piece of work! Thanks a lot for the effort. 
> 
> I'll stay tuned for the V3 implementations
> 
> /Antonio

You're quite welcome, and thank you.  
My primary goal for CFA is to have a complete implementation of The Crossfire Protocol.