ZZ-Cat/CRSFforArduino #153: Connection fails over time

Invertisment, 2025-06-28:
### Is there an existing issue for this bug?

- [x] I have searched the issues and there is no existing issue for this bug.

### What development environment are you using?

Arduino IDE

### What board are you using?

Teensy 4.0

### What part of CRSF for Arduino is this bug related to?

Other (please specify)

### Current behaviour

I'd like to receive some pointers about a bug that I get.

I use a Betafpv Lite ELRS receiver (tower version) and connect it to TX0/RX0 ports in my Teensy 4.0 board (i.e. it's supported).

### Expected behaviour

No tx/rx problem

### Steps to reproduce

At this moment I don't have a way to reliably reproduce it.
I had the issue only three times so far (starting today).

The connection seems to be fine and initially I can use the channels.

But after some time (1-2 minutes) randomly the connection between ELRS receiver and the Teensy stops working.
i.e. the transmitter says it's bound and the receiver chip also has a non-blinking blue light which means it detects the connection with transmitter. 
This means that the connection is alright.

Today I had to pull the battery plug twice while the motors were spinning. 

When the connection problem happened the third time the motors weren't spinning and I noticed that the Teensy microcontroller had a persistent red light (not pin 13 light but the other light) on.
And I also know that the battery was at 11.7V (or 3S with 3.9V per cell). Which means that there should've been enough power to run everything.
When I connected it to the PC then the error disappeared and everything was controllable again.
So something has cleared up the error and the red LED went away without any other reconnections.
I think the connection didn't do a reboot of the board.

Do you have any clues on what could be causing this?

ZZ-Cat, 2025-06-28:
There is a lot of unknowns here. I need you to be more specific.  
For a start, how _exactly_ are you powering this? What's connected to your Teensy 4.0? How is your battery connected to the entire set-up? Where and with what is your receiver connected to—including its power and UART pins? Are all four of your receiver's pins connected to your Teensy 4.0?

Even better... can you provide me with a schematic diagram of your exact setup?

> I don't have a way to reliably reproduce it.

If this _is_ specific to CFA, I _need_ to be able to reproduce it on my end in order to help you.

> When the connection problem happened the third time the motors weren't spinning and I noticed that the Teensy microcontroller had a persistent red light (not pin 13 light but the other light) on. [...]

This suggests to me that it's likely to be a power issue, and I need to rule this out _before_ I can consider whether-or-not CFA is the one at fault here.

Invertisment, 2025-07-01:
I think I didn't run `update` enough. That might be my issue. I'll test more later.

I'm powering the teensy via ESCs that are providing 5V BEC via the VCC pin.

The ELRS receiver is taking this same VCC 5V trace (not from teensy).

TX is connected to RX0 and RX is connected to TX0.

ZZ-Cat, 2025-07-01:
Yeha, because CFA currently doesn't provide its own hardware drivers (and by extension, uses Arduino's `Serial1` UART API), what `update()` is doing is... it polls the buffer for new data, and that's done _per byte._  
Therefore, `update()` MUST be called as fast as possible.

On your receiver's side, it might pay to set the baud rate (I can't remember the setting to do this, off the top of my head, but it's in ExpressLRS' configurator) to 420,000, because I have not yet updated CFA's own baudrate to reflect the correct 416,666 baudrate that CRSF is designed to use.

Invertisment, 2025-07-06:
I'll close this as I don't plan to work on it in the meantime. It's probably a non-issue. I'll reopen if it will come back.

closed 2025-07-06 by Invertisment